nopnop2002/esp-idf-sx127x #2: Any plans to support dio0 and dio1?

specternecter, 2022-02-07:
I'm trying to move over to the idf from Arduino because the esp32c3 usb support is absolutely terrible there. I was using the radiolib arduino library for the sx1276 and it works great. But it doesn't work in the idf obviously. It does have support for dio0 dio1 (and rxen txen pins that a lot of modules have). Are these pins going to be added in the future?

nopnop2002, 2022-02-07:
Is there a sample somewhere that uses DIO0, DIO1, rxen, txen pins?

specternecter, 2022-02-08:
It turns out rxen and txen aren't mentioned in most libraries because it's a switch that turns on the circuit that allows the modified sx127x modules to transmit at 1W and it's not hard do do this manually. This is not an option created by semtech so it's not in the datasheets. It's an aftermarket upgrade essentially and it's in their datasheets (those ebyte modules) but as is typical of datasheets from China it's full of conflicting information that ends up making no sense. My understanding is to transmit txen must be low and rxen high, and the opposite for receive. The radiolib library just allows you to hand over control and not have to worry about it

nopnop2002, 2022-02-08:
To understand these features, I need to get the e32-900m30s.

vogelfreiheit, 2022-05-06:
I will try to help @nopnop2002. This library supports the DIO0 pin as IRQ / RX signaling.

https://github.com/bkgoodman/esp-idf-lora

I'm having issues with it only because TTGO boards share the SPI bus with the sd card.

nopnop2002, 2022-05-08:
>This library supports the DIO0 pin as IRQ / RX signaling.

I couldn't figure it out by looking at your source.

Please tell me a little more detail.

nopnop2002, 2022-05-12:
I want to know how to use DIO0 and DIO1.

If these are for interrupts, then ESP-IDF does not require interrupts.

Unlike the Arduino environment, you can loop for each task.

satyamedh, 2025-02-16:
hey @nopnop2002 , I assume you're asking whats the purpose of the DIO0 and DIO1 pins?

I need to use them as a wake up from sleep, basically ->
1. ESP32 is in sleep to conserve power
2. lora module picks up a tranmission and stores it in the buffer
3. lora module signals DIO0 HIGH(at least thats what I got from the datasheet)
4. ESP32 wakes up and reads the packet, takes appropriate action and returns to sleep

This is how I would like to use it, and I'm not really experienced with ESP-IDF or embedded, so forgive me if I got something wrong

nopnop2002, 2026-09-01:
The latest library can notify the MCU of interrupts.

https://github.com/nopnop2002/esp-idf-sx127x/tree/main/interrupt