nopnop2002/esp-idf-sx127x #3: Which sx127*  module ?

netmonk, 2022-02-15:
Hello, which SX127* module are you using for dev this library ? 
i guess hoperf rfm95 or any board with an esp32 and lora module like Liligo.

Why i ask ? Because im looking for support of EbYTES sx1276/78 modules like E19-868M30S
Those modules are designed with some added pin like TXEN/RXEN that much be respetively HIGH when sending or receiving.
But i never see ESP-IDF library which implement this.

netmonk, 2022-02-15:
I played with two lora lilig lora32 modules and both basic/pingpong are working fine.
Congratulation, now is time to test esp32c3 and Ebyte E19 modules.

netmonk, 2022-02-15:
[image]

netmonk, 2022-02-15:
Funny the second partner seems to have received some other lora packet (may be my gateway ttn) 
[image]

netmonk, 2022-02-15:
Btw, 60 to 70 ms of back to back time is like 16hz!
Expresslrs on same hardware reach easely 100hz up to 250hz on 868/915 
it can reach 500hz packet rate using 2.8ghz lora modules

nopnop2002, 2022-02-15:
>Hello, which SX127* module are you using for dev this library ?

I used these:
https://www.aliexpress.com/item/32911759847.html?spm=a2g0o.9042311.0.0.27424c4dL4D3Cc
https://www.aliexpress.com/item/32962551530.html?spm=a2g0o.9042311.0.0.6aa24c4dHXeafK

netmonk, 2022-02-16:
The EBYTE E28 https://www.ebyte.com/en/product-view-news.html?id=537
is like the Ebyte E28 very small pinout module.
I made an expension board for the E22 Module same can be done for the E28 Module :
[image]

nopnop2002, 2022-02-16:
>I made an expension board for the E22 Modul

I want it.....

netmonk, 2022-02-16:
Send email to netmonk-at-netmonk dot org with your adress i will send some sample. Beware it’s pin labelled for  E22 , but you can solder and use an E28 module too

nopnop2002, 2022-02-16:
Thank you for your offer.

If you sell this on ebay, I will buy it.

netmonk, 2022-02-16:
https://easyeda.com/dpac/e22-carrier-board 
You can find the open design here. My proposition was to send you a vrigin pcb board for free.
I can also send a mounted board (need to have two in order to test new feature such has SF 5 ) far free. :)

nopnop2002, 2022-02-16:
>My proposition was to send you a vrigin pcb board for free.

Really?

netmonk, 2022-02-16:
Yes

nopnop2002, 2022-02-16:
You need Shipping Cost.
Are you OK?

netmonk, 2022-02-16:
I appreciate you worry about that but don't :)

nopnop2002, 2022-02-16:
I sent mail to you.
Thank you again.

netmonk, 2022-02-16:
im not sure you click "send" i received nothing : netmonk@netmonk.org

nopnop2002, 2022-02-16:
I made a mistake.

I sent it again.

netmonk, 2022-02-16:
Great received :)

nopnop2002, 2022-02-16:
When I receive the extension board, I would like to hear your opinion on which communication module it is worth testing.

netmonk, 2022-03-16:
Did you receive anything  by airmail ?

nopnop2002, 2022-03-16:
No.

Probably the influence of Ukraine.

netmonk, 2022-05-06:
Hello did you receive it ?

closed 2022-07-31 by netmonk