nopnop2002/esp-idf-sx127x #5: Conflict with SPI sdcard driver (InstrFetchProhibited)

vogelfreiheit, 2022-05-06:
I've ported part of the IRQ handling code from 
```
    esp_err_t err = ESP_OK;

#ifdef LED_PIN
    gpio_set_direction(LED_PIN, GPIO_MODE_OUTPUT);
#endif

#if CONFIG_I2C_INTERFACE
    int i2c_sda = CONFIG_SDA_GPIO;
    int i2c_sclk = CONFIG_SCL_GPIO;
    int reset_pin = CONFIG_RESET_GPIO;

	ESP_LOGI(TAG, "INTERFACE is i2c");
	ESP_LOGI(TAG, "CONFIG_SDA_GPIO=%d",CONFIG_SDA_GPIO);
	ESP_LOGI(TAG, "CONFIG_SCL_GPIO=%d",CONFIG_SCL_GPIO);
	ESP_LOGI(TAG, "CONFIG_RESET_GPIO=%d",CONFIG_RESET_GPIO);

    i2c_config_t i2c_config = {
		.mode = I2C_MODE_MASTER,
		.sda_io_num = i2c_sda,
		.scl_io_num = i2c_sclk,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = I2C_MASTER_FREQ_HZ
	};

	ESP_ERROR_CHECK(i2c_param_config(I2C_NUM, &i2c_config));
	ESP_ERROR_CHECK(i2c_driver_install(I2C_NUM, I2C_MODE_MASTER, 0, 0, 0));

    if (reset_pin >= 0) {
		//gpio_pad_select_gpio(reset);
		gpio_reset_pin(reset_pin);
		gpio_set_direction(reset_pin, GPIO_MODE_OUTPUT);
		gpio_set_level(reset_pin, 0);
		vTaskDelay(50 / portTICK_PERIOD_MS);
		gpio_set_level(reset_pin, 1);
	}

#ifdef HAS_OLED_DISPLAY
    ssd1306_init(&ssd1306_dev, 128, 64);
    ssd1306_clear_screen(&ssd1306_dev, false);
	ssd1306_contrast(&ssd1306_dev, 0xff);
#endif
#endif // CONFIG_I2C_INTERFACE

#ifdef USE_SDCARD
    spi_bus_config_t bus_cfg = {
        .mosi_io_num = SDCARD_MOSI,
        .miso_io_num = SDCARD_MISO,
        .sclk_io_num = SDCARD_SCK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 4000,
    };
    ret = spi_bus_initialize(host.slot, &bus_cfg, SDSPI_DEFAULT_HOST);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to initialize bus.");
        return ret;
    }
#endif

    err = lora_setup(cfg);
    if (err != ESP_OK)
        ESP_LOGW(TAG, "RF communications not functional. Missing component?");

```

My SD card code is identical to:

vogelfreiheit, 2022-05-06:
An update: `spi: spi_bus_initialize(628): SPI bus already initialized.␛[0` this is possibly related to not using the right bus (I do not know yet how to force use of software SPI bus).

nopnop2002, 2022-05-08:
sdmmc_host_t host initialize.

```
sdmmc_host_t host = SDSPI_HOST_DEFAULT();
```

SDSPI_HOST_DEFAULT () has the following contents.

```
#if CONFIG_IDF_TARGET_ESP32 || CONFIG_IDF_TARGET_ESP32S2
#define SDSPI_DEFAULT_HOST HSPI_HOST
#define SDSPI_DEFAULT_DMA  SDSPI_DEFAULT_HOST
#else
#define SDSPI_DEFAULT_HOST SPI2_HOST
#define SDSPI_DEFAULT_DMA  SPI_DMA_CH_AUTO
#endif

/**
 * @brief Default sdmmc_host_t structure initializer for SD over SPI driver
 *
 * Uses SPI mode and max frequency set to 20MHz
 *
 * 'slot' should be set to an sdspi device initialized by `sdspi_host_init_device()`.
 */
#define SDSPI_HOST_DEFAULT() {\
    .flags = SDMMC_HOST_FLAG_SPI | SDMMC_HOST_FLAG_DEINIT_ARG, \
    .slot = SDSPI_DEFAULT_HOST, \
    .max_freq_khz = SDMMC_FREQ_DEFAULT, \
    .io_voltage = 3.3f, \
    .init = &sdspi_host_init, \
    .set_bus_width = NULL, \
    .get_bus_width = NULL, \
    .set_bus_ddr_mode = NULL, \
    .set_card_clk = &sdspi_host_set_card_clk, \
    .do_transaction = &sdspi_host_do_transaction, \
    .deinit_p = &sdspi_host_remove_device, \
    .io_int_enable = &sdspi_host_io_int_enable, \
    .io_int_wait = &sdspi_host_io_int_wait, \
    .command_timeout_ms = 0, \
}
```

The same SLOT cannot be used for host.slot when initializing SDSPI and SLOT when initializing SPI devices other than SDSPI.

```
    ret = spi_bus_initialize(host.slot, &bus_cfg, SDSPI_DEFAULT_HOST);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to initialize bus.");
        return ret;
    }
```

nopnop2002, 2022-05-08:
>I've ported part of the IRQ handling code from https://github.com/bkgoodman/esp-idf-lora, to your implementation. 

I want to know more details.

nopnop2002, 2022-05-09:
I found this code no longer meaningful:
ESP32 has SPI2_HOST/SPI3_HOST.
```
#ifdef CONFIG_IDF_TARGET_ESP32
#define LCD_HOST HSPI_HOST
#elif defined CONFIG_IDF_TARGET_ESP32S2
#define LCD_HOST SPI2_HOST
#elif defined CONFIG_IDF_TARGET_ESP32S3
#define LCD_HOST SPI2_HOST
#elif defined CONFIG_IDF_TARGET_ESP32C3
#define LCD_HOST SPI2_HOST
#endif
```

I am considering making it possible to select SPI2_HOST / SPI3_HOST.

nopnop2002, 2022-05-12:
Enabled to select SPI2_HOST / SPI3_HOST
[image]

closed 2022-08-01 by nopnop2002