nopnop2002/esp-idf-sx127x #7: stack overflow on esp32-c3

netmonk, 2022-07-31:
creating task with 1024*2 stack size is too small and therefore lead to carsh loop with stack overflow when lauching either task_rx or task_tx in basic exemple. 

 ```
#if CONFIG_SENDER                                                                                                                                                                                          
          xTaskCreate(&task_tx, "task_tx", 1024*4, NULL, 5, NULL);                                                                                                                                           
#endif                                                                                                                                                                                                     
#if CONFIG_RECEIVER                                                                                                                                                                                                  
          xTaskCreate(&task_rx, "task_rx", 1024*4, NULL, 5, NULL);                
```
Solution is to increase stack size at task creation.

nopnop2002, 2022-08-01:
>creating task with 1024*2 stack size is too small

I don't have such code in my repository.

```
./ping-pong/main/main.c:        xTaskCreate(&task_primary, "task_primary", 1024*3, NULL, 5, NULL);
./ping-pong/main/main.c:        xTaskCreate(&task_seconfary, "task_secondary", 1024*3, NULL, 5, NULL);
./basic/main/main.c:    xTaskCreate(&task_tx, "task_tx", 1024*3, NULL, 5, NULL);
./basic/main/main.c:    xTaskCreate(&task_rx, "task_rx", 1024*3, NULL, 5, NULL);
```

closed 2022-08-01 by nopnop2002

netmonk, 2022-08-01:
Exact, it was on basic.c and I didn’t pull since 3 of may. All apologise.