nopnop2002/esp-idf-sx127x #8: Is possible communicate p2p?

julioj4pa, 2022-12-13:
Hello, awesome code. Thank you for sharing.
I'm already using Wifi of ESP-IDF to communication.
I'm so newbie, I read library but I can't find how to communicate peer to peer... Is it possible? If is how can I do it?
I'm really thankful for attention.

nopnop2002, 2022-12-13:
P2P is possible.
Requires 2 ESP32s and 2 LoRa modules.
Please try this.

https://github.com/nopnop2002/esp-idf-sx127x/tree/main/basic

julioj4pa, 2022-12-14:
I have multiple nodes in network. I want to address each node like multicast and send message to specific node.
How to put address each one to direct communication?
If not I'll send messages as broadcast and process message at each node...

nopnop2002, 2022-12-14:
You can communicate only with the other party whose SyncWord matches.
```
lora_set_sync_word(int sw)
```

closed 2023-07-20 by nopnop2002