nopnop2002/esp-idf-sx127x #9: ESP-IDF Heap Poisoning

manuellapanza, 2023-01-13:
hi, I'm testing this library but I'm having a specific problem of heap poisoning in void function **_lora_read_reg_buffer._** Every time a packet is received the esp idf terminal returns this error:

```
assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)

Backtrace:0x40081d96:0x3ffbea200x400888a5:0x3ffbea40 0x4008e819:0x3ffbea60 0x4008e437:0x3ffbeb80 0x40082006:0x3ffbeba0 0x400825a7:0x3ffbebc0 0x40082621:0x3ffbebf0 0x400f4581:0x3ffbec10 0x400f4972:0x3ffbec60 0x400d88cd:0x3ffbec80 0x400d897a:0x3ffbeda0 0x400d839a:0x3ffbedc0 0x400d83f6:0x3ffbede0 0x4008b975:0x3ffbee00
0x40081d96: panic_abort at C:/Users/esp/esp-idf/components/esp_system/panic.c:402

0x400888a5: esp_system_abort at C:/Users/esp/esp-idf/components/esp_system/esp_system.c:128

0x4008e819: __assert_func at C:/Users/esp/esp-idf/components/newlib/assert.c:85

0x4008e437: multi_heap_free at C:/Users/esp/esp-idf/components/heap/multi_heap_poisoning.c:253 (discriminator 1)

0x40082006: heap_caps_free at C:/Users/esp/esp-idf/components/heap/heap_caps.c:340

0x400825a7: trace_free at C:/Users/esp/esp-idf/components/heap/include/heap_trace.inc:117

0x40082621: __wrap_free at C:/Users/esp/esp-idf/components/heap/include/heap_trace.inc:162

0x400f4581: lora_read_reg_buffer at c:\users\documents\teste\anthonomus-esp-trap\firmware\build/../lib/LoRa/src/LoRa.c:131

0x400f4972: lora_receive_packet at c:\users\documents\teste\anthonomus-esp-trap\firmware\build/../lib/LoRa/src/LoRa.c:655 
```

I tried commenting out this line from **_free(in)_** l.131. It works, but doing so leaks memory in the code. I've already tried to put some prints in the function to see if I could find the reason for this heap poisoning error, but I couldn't. Could you help me with some solution to this problem please?

nopnop2002, 2023-01-13:
Tell me the version of ESP-IDF you are using

manuplf, 2023-01-14:
I'm using version 4.4

nopnop2002, 2023-01-14:
I tried it with ESP-IDF V4.4, but no problem occurs.
My board is ESP32.

```
I (357) task_rx: Start
I (3977) lora_read_reg_buffer: start
I (3977) lora_read_reg_buffer: end
I (3977) task_rx: 19 byte packet received:[Hello World!! 14172]
I (9037) lora_read_reg_buffer: start
I (9037) lora_read_reg_buffer: end
I (9037) task_rx: 19 byte packet received:[Hello World!! 14678]
```

my environment:
```
$ idf.py --version
ESP-IDF v4.4.3-347-g9ee3c8337d
```

Pls change this and try again.

https://github.com/nopnop2002/esp-idf-sx127x/blob/main/components/lora/lora.c#L88
#define BUFFER_IO 0

manuellapanza, 2023-01-14:
I tested it with the example and it really works. I'm using the library for a larger project with a slightly larger amount of data when sending packets, I don't know if that could be influencing this error... 
Here is my version:
```
> idf.py --version
ESP-IDF v4.4
```
I think it was supposed to be working haha. The only problem is with the **lora_read_reg_buffer**, even the **lora_write_reg_buffer** works correctly

manuellapanza, 2023-01-14:
And I also tested it in my project by setting BUFFERIO to 0 and it works, but I also wanted to know because in the comment it says that using this is a little faster, how much faster would it be? Does it make that much of a difference? Because this agility would be great for my project

nopnop2002, 2023-01-14:
>ESP-IDF v4.4

It is too older version.
You can update below.

```
git clone -b release/v4.4 --recursive https://github.com/espressif/esp-idf.git
```

nopnop2002, 2023-01-21:
I will close this issue.

You can RE-OPEN at any time.

closed 2023-01-21 by nopnop2002