nopnop2002/esp-idf-sx127x #10: how to added fixed address and channel

cyberumut, 2023-03-17:
how to added fixed address and channel ? 

i need address & channel 
as in the picture
[image]

nopnop2002, 2023-03-17:
Where is your drawing from?

cyberumut, 2023-03-17:
ebyte modules
Isn't this header structure global?
targetAddr | TargetChannel | Data 
how can i  apply this on sx1278
> Where is your drawing from?
> 
> I can't find it in the SX1276/77/78/79 datasheet.

nopnop2002, 2023-03-17:
Where can I get your documentation?

tell me the url.

cyberumut, 2023-03-17:
https://www.cdebyte.com/products/E22-400T30D/4#Downloads

check it manual .pdf 
> LoRa Packet Structure

nopnop2002, 2023-03-17:
EBYTE's E22 module uses different frequencies depending on the channel.

[image]

you can use any frequency you like in menuconfig

[image]

cyberumut, 2023-03-17:
module in hand is sx1278 ,not ebyte module
I want to define address and channel on sx1278
how can i do ? 

> EBYTE's E22 module uses different frequencies depending on the channel.
> 
> [image]
> 
> you can use any frequency you like in menuconfig
> 
> [image]

nopnop2002, 2023-03-17:
This diagram you showed is from EBYTE.

[image]

You're using EBYTE's SX1278, right?

cyberumut, 2023-03-17:
no i dont using eybte module .
I just want to set up addressing and channel structure like it

>

nopnop2002, 2023-03-17:
The LoRa specification does not have channels or addresses.

EBYTE modules channel is same as frequency

cyberumut, 2023-03-17:
If we add such a structure to the header. I think it can work if we parse the header on the other device or 
i selected channel 4  
Frequency= 410.125 + CH*1M    
410.125+ 4*1M
it needs to be calculated
[image]

> The LoRa specification does not have channels or addresses.
> 
> EBYTE modules channel is same as frequency
> 
> For modules other than EBYTE, change the frequency if you use the concept of channels.

nopnop2002, 2023-03-17:
I do not understand your content.

The LoRa packet structure is the following structure and Header is optional.

[image]

nopnop2002, 2023-03-17:
EBYTE LoRa modules and non-EBYTE LoRa modules may not be able to communicate with each other.

This is because EBYTE has the concept of addresses, but LoRa modules other than EBYTE do not.

Presumably the EBYTE address is sent as part of the LoRa payload.
This is an EBYTE-only payload.

EBYTE-specific payloads cannot be interpreted correctly by non-EBYTE LoRa modules.

cyberumut, 2023-03-17:
i got it now thank u bro
> EBYTE LoRa modules and non-EBYTE LoRa modules may not be able to communicate with each other.
> 
> This is because EBYTE has the concept of addresses, but LoRa modules other than EBYTE do not.
> 
> Presumably the EBYTE address is sent as part of the LoRa payload. This is an EBYTE-specific payload.
> 
> EBYTE-specific payloads cannot be interpreted correctly by non-EBYTE LoRa modules.

closed 2023-03-17 by nopnop2002